PyLabRobot/pylabrobot #5: assign_resource missing from LiquidHandler

marcosfelt, 2022-10-01:
It seems the `assign_resource` method is missing from the `LiquidHandler` class. I get the following error:

```bash
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In [7], line 1
----> 1 lh.assign_resource(tip_car, rails=1, replace=True)

AttributeError: 'LiquidHandler' object has no attribute 'assign_resource'
```

Was this unintentionally deleted in a recent commit?

rickwierenga, 2022-10-01:
This has been replaced with `lh.deck.assign_child_resource`.

Is the code above from one of the official samples? (except `basic.ipynb`, which I'm updating today when I get access to the robot)

marcosfelt, 2022-10-01:
Perfect! It's actually in `using_the_simulator.ipynb`.  I have a couple things I noticed that could make that notebook clearer. Should I send in a PR?

rickwierenga, 2022-10-01:
That’d be great! Thanks in advance.

closed 2022-10-02 by rickwierenga