PyLabRobot/pylabrobot #13: Issues in the docs

naikymen, 2023-01-08:
Hello!

I've completed my Klipper setup for a custom pipetting robot, and wanted to write a backend for it.

I though I'd start by running the examples from [the docs](https://docs.pylabrobot.org/) first.

I'll leave over here some of the issues I faced by following the instructions over there.

## Installing PLR

> From: https://docs.pylabrobot.org/installation.html#installing-pylabrobot

The command `python -m virtualenv env` did not work for me.

Instead I needed to use:

```bash
python -m venv env
```

## Installing from source

> From: https://docs.pylabrobot.org/installation.html#from-source

I was encouraged by the docs to install from source, because _this is particularly useful if you want to contribute to the project_.

However, `pip install -e '.[dev]'` fails due to a version incompatibility between Jupyter and Opentrons.

```
jupyterlab-server 2.18.0 requires jsonschema>=4.17.3

opentrons 6.2.0 requires jsonschema==3.0.2
```

**Q1**: why is the opentrons module required?

This contrasts with my expectation of a hardware-agnostic framework.

From now on, I used PLR installed from `pip` in a new virtual environment, which seems to work.

## Using the simulator

> From: https://docs.pylabrobot.org/using-the-simulator.html

I wanted to use the simulator, and found some import errors. I imagined I could install the missing PLR resources using `pip install pylabrobot[simulation]`.

After that, I opened a python console and ran:

```python
from pylabrobot.liquid_handling import LiquidHandler
```

Which failed with the error: `ModuleNotFoundError: No module named 'typing_extensions'`.

I traced the error a bit. Something must be off with the detection of python version.

After installing the `typing_extensions` module manually, I ran the PLR import line again, and found another missing module: `NameError: name 'LabwareDefinition' is not defined`. This one should have been imported from `opentrons_shared_data.labware.dev_types`.

See **Q1** above.

Unsure about how to get past the import errors, I resorted to run: `pip install pylabrobot[all]`. This downloaded 17 different versions of pyhamilton. I noticed that the `opentrons` option is not listed in the [relevant section](https://docs.pylabrobot.org/installation.html#using-pip) of the docs.

I ran the PLR import line again, which finally worked! :partying_face: 

I'll continue adding my notes tomorrow.

Cheers!

rickwierenga, 2023-01-08:
Hi,

First off, thank you for the perseverance and writing such a detailed issue. I'm glad you got it working!

> The command python -m virtualenv env did not work for me.

[virtualenv](https://virtualenv.pypa.io/en/latest/) is a library that needs to be installed separately whereas `venv` is included in the standard library. The page I linked to summarizes some of the differences. https://github.com/PyLabRobot/pylabrobot/commit/9de3c79c9c58f833785c459d8ae2e8b429c3cb69 should make things clearer.

> Q1: why is the opentrons module required?

The `dev` target includes all dependencies that are be needed for PyLabRobot development. PyLabRobot includes an Opentrons integration which depends on the opentrons python library. Although you will probably not be making changes here, the library is required for running automated tests, for example.

> From now on, I used PLR installed from pip in a new virtual environment, which seems to work.

Note that the pip version is actually slightly outdated. It also does not include some files that are needed for development, so I'd highly recommend you fork the repo and install from source. Here's a guide on that: https://docs.pylabrobot.org/how-to-open-source.html.

> Which failed with the error: ModuleNotFoundError: No module named 'typing_extensions'.

I forgot to add this as a dependency. Should be fixed by https://github.com/PyLabRobot/pylabrobot/commit/8c2281024693594584c8b529afdbf9684dd1f75e.

> I ran the PLR import line again, and found another missing module: NameError: name 'LabwareDefinition' is not defined. This one should have been imported from opentrons_shared_data.labware.dev_types.

In this case, PyLabRobot required Opentrons even if it was not actually used. That's a bug, which should be fixed by https://github.com/PyLabRobot/pylabrobot/commit/ca397e87caf757fc4258ca1ab19b64d5183fd53a.

> This downloaded 17 different versions of pyhamilton.

What do you mean by 17 different versions? PyLabRobot should only require the latest version on PyPI.

> I noticed that the opentrons option is not listed in the [relevant section](https://docs.pylabrobot.org/installation.html#using-pip) of the docs.

Should be addressed by https://github.com/PyLabRobot/pylabrobot/commit/2a139e60f6e335dac3bd014e3a281ea49be2b760.

Thanks again for reporting all of this!

naikymen, 2023-01-08:
Hi Rick! Awesome, thanks for the quick response and fixes :)

> [virtualenv](https://virtualenv.pypa.io/en/latest/) is a library that needs to be installed separately whereas venv is included in the standard library.

Thanks :) glad to learn something new.

> The dev target includes all dependencies that are be needed for PyLabRobot development. 

Makes sense, thanks!

> What do you mean by 17 different versions? PyLabRobot should only require the latest version on PyPI.

The following pyhamilton files were downloaded yesterday (note: they now appear as cached):

```
Collecting pyhamilton
  Using cached pyhamilton-1.491.4-py3-none-any.whl (15.9 MB)
  Using cached pyhamilton-1.491.3-py3-none-any.whl (15.9 MB)
  Using cached pyhamilton-1.491.2-py3-none-any.whl (15.9 MB)
  Using cached pyhamilton-1.491.1-py3-none-any.whl (15.9 MB)
  Using cached pyhamilton-1.490.1-py3-none-any.whl (15.9 MB)
  Using cached pyhamilton-1.490-py3-none-any.whl (15.8 MB)
  Using cached pyhamilton-1.489-py3-none-any.whl (15.8 MB)
  Using cached pyhamilton-1.487-py3-none-any.whl (15.7 MB)
  Using cached pyhamilton-1.486-py3-none-any.whl (5.8 MB)
  Using cached pyhamilton-1.485-py3-none-any.whl (5.8 MB)
  Using cached pyhamilton-1.483-py3-none-any.whl (5.8 MB)
  Using cached pyhamilton-1.49.1-py3-none-any.whl (15.9 MB)
  Using cached pyhamilton-1.48-py3-none-any.whl (5.8 MB)
  Using cached pyhamilton-1.47-py3-none-any.whl (3.7 MB)
  Using cached pyhamilton-1.46-py3-none-any.whl (3.7 MB)
  Using cached pyhamilton-1.45-py3-none-any.whl (909 kB)
  Using cached pyhamilton-1.44-py3-none-any.whl (6.4 MB)
```

I'm not yet familiar with pip packaging. Is this expected?

naikymen, 2023-01-08:
Add.: I just retried `pip install pylabrobot[all]` and got the PIP error related to jsonchema versions:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
jupyterlab-server 2.18.0 requires jsonschema>=4.17.3, but you have jsonschema 3.0.2 which is incompatible.
jupyter-events 0.5.0 requires jsonschema[format-nongpl]>=4.3.0, but you have jsonschema 3.0.2 which is incompatible.
Successfully installed jsonschema-3.0.2 pyhamilton-1.44
```

naikymen, 2023-01-08:
PS: I think that the jsonchema conflict is not on PLR. I'm pretty sure I forgot to activate the vritual environment before running it. Sorry!

rickwierenga, 2023-01-08:
> I'm not yet familiar with pip packaging. Is this expected?

I don't think so, but I will ask Stefan, because he manages PyHamilton.

naikymen, 2023-01-08:
# ... continuing the reports

> From: https://docs.pylabrobot.org/using-the-simulator.html

## lh.setup

While running `lh.setup()`:

- A new browser tab opened, showing the "PyLabRobot Simulator (v0.1)" in a "Connected" status.
- The following error appeared:

```
Simulation server started at http://127.0.0.1:2121/
File server started at http://127.0.0.1:1337./ Open this URL in your browser.
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[5], line 1
----> 1 lh.setup()

File ~/Projects/GOSH/gosh-col-dev/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:112, in LiquidHandler.setup(self)
    109 self.head = {c: ChannelTipTracker() for c in range(self.backend.num_channels)}
    111 for resource in self.deck.children:
--> 112   self.resource_assigned_callback(resource)

File ~/Projects/GOSH/gosh-col-dev/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:148, in LiquidHandler.resource_assigned_callback(self, resource)
    147 def resource_assigned_callback(self, resource: Resource):
--> 148   self.backend.assigned_resource_callback(resource)

File ~/Projects/GOSH/gosh-col-dev/pylabrobot/pylabrobot/liquid_handling/backends/serializing_backend.py:44, in SerializingBackend.assigned_resource_callback(self, resource)
     43 def assigned_resource_callback(self, resource: Resource):
---> 44   self.send_command(command="resource_assigned", data=dict(resource=resource.serialize(),
     45     parent_name=(resource.parent.name if resource.parent else None)))

File ~/Projects/GOSH/gosh-col-dev/pylabrobot/pylabrobot/liquid_handling/backends/websocket.py:191, in WebSocketBackend.send_command(self, command, data, wait_for_response)
    189 # Run and save if the websocket connection has been established, otherwise just save.
    190 if wait_for_response and not self.has_connection():
--> 191   raise ValueError("Cannot wait for response when no websocket connection is established.")
    193 if self.has_connection():
    194   asyncio.run_coroutine_threadsafe(self.websocket.send(serialized_data), self.loop)

ValueError: Cannot wait for response when no websocket connection is established.
```

Is this expected?

I then followed steps up to "liquid handling" where other errors popped up.

I'm not sure how to proceed, but it's enough to start working.

rickwierenga, 2023-01-08:
> Is this expected?

Actually yes, but it's not desired :)

I hope https://github.com/PyLabRobot/pylabrobot/commit/f961ab15e623688b3bf647551101151c44d5997e fixes it.

rickwierenga, 2023-09-21:
Hi Nicolás, did you ever get this to work? How's your project going?

naikymen, 2023-09-21:
Hi Rick! Glad to hear from you. I have not tested the docs further since last time, fell free to close this one. :smile: 

The project is going really well, and [about a month ago](https://gitlab.com/pipettin-bot/forks/pylabrobot/-/commits/pipetting?ref_type=heads) I managed to get PLR to move our robot to a resource's position, but that's about it for now.

There has been a lot of refactoring in both hardware and software, which has put off work on PLR. What is really missing to really test the backend is an adapter, that translates our "Workspace" definition to a JSON that PLR can de-serialize into a Deck.

If you are interested in using our GUI to define PLR Decks, and have time and will to lend a hand, I would really appreciate it.

I can't believe that 9 months have passed since my last comment, but I've not abandoned the project at all. :)

Best!

rickwierenga, 2023-09-25:
Nice to hear from you too! That's good progress!

Having your GUI be interoperable with PLR sounds extremely useful! Unfortunately, I'm afraid that doing my last semester of classes (plus those I missed last year to work on PLR) will make it hard for me to find significant time to dedicate to other projects in the near future...

In the meantime, are you able to define your deck through the PLR Resource model? I think subclassing Deck should give you most of what you need. Granted, it's not as user friendly as your GUI, but at least you'll be able to hook into the PLR ecosystem that way. Happy to help!

rickwierenga, 2024-06-24:
I think this issue is resolved. Can I close this?

(Let's continue discussing your Klipper backend in a new issue!)

naikymen, 2024-06-25:
Hi Rick! Yes :)

closed 2024-06-25 by naikymen