PyLabRobot/pylabrobot #21: Use OT-2 Scripts without Opentrons App

noparis, 2023-02-08:
Hello,
I try to use the PyLabRobot library for Opentron (I used the script some modification as in the video demo https://www.youtube.com/watch?v=8YMcIpf-u1s), but I didn’t succeed to use Opentron with PyLabRobot. 
Do you have others notebook using the Opentron with PyLabRobot ?
Thanks for you help.

rickwierenga, 2023-02-08:
What error are you getting and what modifications did you make?

noparis, 2023-02-08:
Thank you for your quick answer.

I changed the following import because it didn’t find the module :
from pylabrobot.liquid_handing.resources.opentrons import OTDeck => from pylabrobot.resources.opentrons import OTDeck.

Then I get a first error from the following command :
>>> await lh.setup()
It return : AttributeError: type object 'OTDeck' has no attribute 'children'

The second error was with the two commands :
>>> lh.deck.assign_child_at_slot(tips, slot =4)
>>> lh.deck.assign_child_resource(tips, slot =4).
It return : missing 1 argument this is 'self'.

It’s strange because when I use PyLabRobot simulator, the commands “lh.deck.assign_child_at_slot(tips, slot =4)”  and “await lh.setup()” run.

rickwierenga, 2023-02-08:
You probably passed the class `OTDeck` to `Liquidhandler` (`Liquidhandler(backend=OTBackend(...), deck=OTDeck)`), which should be an instance: `OTDeck()`, with parentheses.

noparis, 2023-02-09:
Thank you for your quick answer.

closed 2023-02-09 by noparis