PyLabRobot/pylabrobot #22: Problem with pick_up_tips() function

noparis, 2023-02-15:
Hello,

I have a problem with the pick_up_tips() command, when I try to use Pylabrobot for Opentrons.

I have put the commands I used and the error message :

```
from pylabrobot.liquid_handling import LiquidHandler
from pylabrobot.liquid_handling.backends.opentrons_backend import OpentronsBackend
from pylabrobot.resources.opentrons import OTDeck

ot2 = OpentronsBackend(host=hostname, port= 31950)
lh = LiquidHandler(backend= ot2, deck=OTDeck())

await lh.setup()

from pylabrobot.resources.opentrons.tip_racks import opentrons_96_filtertiprack_20ul
from pylabrobot.resources.opentrons.plates import corning_96_wellplate_360ul_flat
tips = opentrons_96_filtertiprack_20ul(name="tip_rack")
plate = corning_96_wellplate_360ul_flat(name="plate")

lh.deck.assign_child_at_slot(tips, slot = 4)
lh.deck.assign_child_at_slot(plate, slot = 5)

plate.set_well_volumes([[20]*12]*8) 

await lh.pick_up_tips(tips["A1"])
```

I got the following error message : 
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[16], line 1
---> 1 await lh.pick_up_tips(tips["A1"])
     2 await lh.aspirate(plate["H6"], vols=10)
     3 await lh.dispense(plate["H5"], vols=10)

File ~/clone_pylab_robot/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:69, in need_setup_finished.<locals>.wrapper(self, *args, **kwargs)
     67 if not self.setup_finished:
     68   raise RuntimeError("The setup has not finished. See `LiquidHandler.setup`.")
---> 69 await func(self, *args, **kwargs)

File ~/clone_pylab_robot/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:350, in LiquidHandler.pick_up_tips(self, tip_spots, use_channels, offsets, **backend_kwargs)
    347   del backend_kwargs[extra]
    349 try:
--> 350   await self.backend.pick_up_tips(ops=pickups, use_channels=use_channels, **backend_kwargs)
    351 except:
    352   for channel, op in zip(use_channels, pickups):
...
---> 40   command_id = f(*args, **kwargs)
     42   end = datetime.datetime.now() + datetime.timedelta(seconds=timeout)
     43   while datetime.datetime.now() < end:

TypeError: pick_up_tip() got an unexpected keyword argument 'offset_x'
```

Thanks for you help.

rickwierenga, 2023-02-15:
I think this is the result of the [`opentrons-http-api-client`](https://github.com/rickwierenga/opentrons-python-api) dependency being outdated. I just pushed a new version to PyPI, so if you reinstall dependencies this should be fixed.

noparis, 2023-02-15:
I did git pull.
When I test with the Opentrons simulation, it works.
I don't have possibility to use Opentrons at the moment, but I will give you a feedback as soon as possible.

Thank you for your quick answer.

rickwierenga, 2023-02-15:
This issue was with an external library, not PyLabRobot, so `git pull` is not sufficient. You should run the `pip install opentrons-http-api-client --upgrade` command to update the dependency.

If you cloned that dependency from git and the issue persists, the error originated elsewhere because I did not update that code.

noparis, 2023-02-15:
The problem is resolved (pull + pip install opentrons-http-api-client --upgrade + restart jupyter-notebook).
Thank you for your help.

closed 2023-02-15 by rickwierenga