PyLabRobot/pylabrobot #23: Problem with the aspirate() function

noparis, 2023-02-15:
Hello,

I have a problem with the aspirate() command, when I try to use Pylabrobot for Opentrons.
But when I use PyLabRobot for Opentrons simulation, it works, I don't get any error message.
I use p20, and she is detected.

I have put the commands I used and the error message : 

```
from pylabrobot.liquid_handling import LiquidHandler
from pylabrobot.liquid_handling.backends.opentrons_backend import OpentronsBackend
from pylabrobot.resources.opentrons import OTDeck

ot2 = OpentronsBackend(host=hostname, port= 31950)
lh = LiquidHandler(backend= ot2, deck=OTDeck())

await lh.setup()

from pylabrobot.resources.opentrons.tip_racks import opentrons_96_filtertiprack_20ul
from pylabrobot.resources.opentrons.plates import corning_96_wellplate_360ul_flat
tips = opentrons_96_filtertiprack_20ul(name="tip_rack")
plate = corning_96_wellplate_360ul_flat(name="plate")

lh.deck.assign_child_at_slot(tips, slot = 4)
lh.deck.assign_child_at_slot(plate, slot = 5)

plate.set_well_volumes([[20]*12]*8) 

await lh.pick_up_tips(tips["A1"])

await lh.aspirate(plate["H6"], vols=5)
```

I got the following error message : 
```
UnexpectedProtocolError: Cannot aspirate more than pipette max volume
```

Thanks for you help.

thiyagarajan17, 2023-02-15:
assign the issue would love to work on it..

rickwierenga, 2023-02-15:
Thanks! Let me know if I can be of any help!

thiyagarajan17, 2023-02-15:
could you check the vol for pylabrobots.. how it handles exceptions when pip exceeds vol

rickwierenga, 2023-02-15:
This error originated in the Opentrons; it's not a PyLabRobot error. (`UnexpectedProtocolError` is not a PyLabRobot error)

In [another issue](https://github.com/PyLabRobot/pylabrobot/issues/22), you mentioned you are having trouble picking up tips. My guess is no tip was picked up, so Opentrons sees the channel as having a capacity of 0.

Could you see if the answer I wrote there works? If this does not work, could you post the error trace?

> how it handles exceptions when pip exceeds vol

See this: https://github.com/rickwierenga/opentrons-python-api/blob/bfe0398237c7b7dc4f88d1f82cdadb1aa9ecaae2/ot_api/decorators.py#L35

noparis, 2023-02-15:
After doing (git pull, pip install opentrons-http-api-client -- update).
I still get the same error message.

The P20 moves to the plate location and descends.

The full message : 
```
---------------------------------------------------------------------------
UnexpectedProtocolError                   Traceback (most recent call last)
Cell In[16], line 1
----> 1 await lh.aspirate(plate["H6"], vols=5)

File ~/clone_pylab_robot/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:69, in need_setup_finished.<locals>.wrapper(self, *args, **kwargs)
     67 if not self.setup_finished:
     68   raise RuntimeError("The setup has not finished. See `LiquidHandler.setup`.")
---> 69 await func(self, *args, **kwargs)

File ~/clone_pylab_robot/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:650, in LiquidHandler.aspirate(self, resources, vols, use_channels, flow_rates, end_delay, offsets, liquid_height, **backend_kwargs)
    647   del backend_kwargs[extra]
    649 try:
--> 650   await self.backend.aspirate(ops=aspirations, use_channels=use_channels, **backend_kwargs)
    651 except:
    652   for op in aspirations:

File ~/clone_pylab_robot/pylabrobot/pylabrobot/liquid_handling/backends/opentrons_backend.py:406, in OpentronsBackend.aspirate(self, ops, use_channels)
    403 else:
    404   offset_x = offset_y = offset_z = 0
--> 406 ot_api.lh.aspirate(labware_id, well_name=op.resource.name, pipette_id=pipette_id,
    407   volume=volume, flow_rate=flow_rate, offset_x=offset_x, offset_y=offset_y, offset_z=offset_z)

File ~/miniconda3/envs/py_3_9/lib/python3.9/site-packages/ot_api/decorators.py:23, in request_with_run_id.<locals>.wrapper(*args, **kwargs)
     20   kwargs["run_id"] = run_id
...
---> 49   raise error_class(error_data["detail"])
     50 elif result["data"]["status"] in ["queued", "running"]:
     51   continue

UnexpectedProtocolError: Cannot aspirate more than pipette max volume
```

rickwierenga, 2023-02-15:
This error must come from the machine itself. Did it pick up the right tip?

You may also need to start a new session using `lh.setup()` (Opentrons has its own volume tracker, which can raise weird errors sometimes).

noparis, 2023-02-16:
Yes, the robot pick up the tips.
I have done it several times (restart, restart + lh.setup(), lh.setup()), and I always get the same error message.

noparis, 2023-02-24:
Hello,

I tried with the last update of Opentrons, and I always get the same error message.

Thanks for you help.

rickwierenga, 2023-02-24:
Are you able to discard the tips? The OT should throw an error if it did not register the pickup.

If that works, could you capture and share the HTTP data using something like [mitmproxy](https://mitmproxy.org)? Use `mitmproxy -p 5005 --mode reverse:http://<OT IP>:31950`, and set the host and port to `0.0.0.0` and `5005` in PLR.

noparis, 2023-03-01:
Hello,

This is what I get after sending the aspirate commande : 

[capture_ecran_HTTP_data.pdf](https://github.com/PyLabRobot/pylabrobot/files/10861141/capture_ecran_HTTP_data.pdf)

rickwierenga, 2023-03-01:
Could you press enter on the command that's being sent to show the JSON data? (the `POST` request)

Also, you can use `-w <file>` to write all output to a file.

noparis, 2023-03-01:
[data_http_1_03.zip](https://github.com/PyLabRobot/pylabrobot/files/10862751/data_http_1_03.zip)

Thanks for you help.

rickwierenga, 2023-03-19:
Unfortunately mitmproxy can't open that file for me. I'm using `mitmproxy==9.0.1`.

rickwierenga, 2023-03-20:
Can you open the file with `mitmproxy -r <path>`?

noparis, 2023-03-20:
With this file ?
[data_20_03_2023.zip](https://github.com/PyLabRobot/pylabrobot/files/11019197/data_20_03_2023.zip)

SyntaxColoring, 2023-03-20:
👋 Hello from Opentrons!

I suspect this has to do with the tip rack's labware definition, which gets `POST`ed in request 80. It defines each "well" with this:

```
"totalLiquidVolume": 0.0
```

But [per our labware definition schema](https://github.com/Opentrons/opentrons/blob/055fafaadd18791fbe52413162ac6638baa360f6/shared-data/labware/schemas/2.json#L239), this should probably be `20`, since the source loads an `opentrons_96_filtertiprack_20ul`.

The labware definition [looks correct on our side](https://github.com/Opentrons/opentrons/blob/055fafaadd18791fbe52413162ac6638baa360f6/shared-data/labware/definitions/2/opentrons_96_filtertiprack_20ul/1.json#L37). I'm not familiar with PyLabRobot's internals, but it looks like this is somehow getting lost in translation? @rickwierenga

rickwierenga, 2023-03-20:
@SyntaxColoring Thanks for the quick reply! That makes sense. I just pushed a commit to fix that. (https://github.com/PyLabRobot/pylabrobot/commit/811a4321dfc6cf3e62ab772bf853996628fc462c)

@noparis, could you see if pulling helps?

rickwierenga, 2023-03-21:
Great!

Thanks @noparis for reporting and @SyntaxColoring for the help!

closed 2023-03-21 by rickwierenga