PyLabRobot/pylabrobot #24: Code runs on Opentron but not simulator

JamesBagley, 2023-02-17:
Hi,
I was trying to use the simulator to catch simple errors in code before handing it over to our opentron to run in "meatspace" but it seems like the behavior of the simulator doesn't exactly match that of the opentron, possibly as it's designed not to be specific to one machine?  A few errors crop up when run but I think 
"ValueError("Must specify at least one channel to pick up tips with.")"
is the root issue, the opentron has only one pipette (a single channel) that is able to pick up the tips so it doesn't need any additional info, whereas I'm guessing the simulator does? 

Currently running on PyLabRobot 0.1.4

Thanks!

```
from pylabrobot.liquid_handling import LiquidHandler
from pylabrobot.liquid_handling.backends.opentrons_backend import OpentronsBackend 
from pylabrobot.liquid_handling.backends.simulation.simulator_backend import SimulatorBackend 
from pylabrobot.liquid_handling.resources.opentrons import OTDeck

from pylabrobot.liquid_handling.resources.opentrons import (
    opentrons_96_tiprack_20ul    
)
from pylabrobot.liquid_handling.resources.corning_costar.plates import  (
    Cos_96_Rd
)

host_ip = ''
def prep_deck(device):
    Source_plate_type = Cos_96_Rd
    Destination_plate_type = Cos_96_Rd
    
    tip_rack = opentrons_96_tiprack_20ul(name="tip_rack")
    source_plate = Source_plate_type(name="source_plate")
    destination_plate = Destination_plate_type(name="destination_plate")
    
    device.deck.assign_child_resource(tip_rack, slot=1)
    device.deck.assign_child_resource(source_plate, slot=2)
    device.deck.assign_child_resource(destination_plate, slot=3)
    
def offset_explore(device):
    tip_rack: TipRack = device.get_resource("tip_rack")
    source_plate: Plate = device.get_resource("source_plate")
    destination_plate: Plate = device.get_resource("destination_plate")

    tip_source = "A1"
    device.pick_up_tips(tip_rack[tip_source])
    for i in range(0,10):
        device.aspirate(source_plate["A1"], vols=[1], offsets_z=[-1*i/5])
    device.discard_tips(tip_rack[tip_source])

backend = OpentronsBackend(host=host_ip, port=31950) 
simulator = SimulatorBackend() 

lh = LiquidHandler(backend=backend, deck=OTDeck())
sim = LiquidHandler(backend=simulator, deck=OTDeck())

for device in [sim,lh]:
    device.setup()
    prep_deck(device)
    offset_explore(device)
    device.stop()
```

rickwierenga, 2023-02-24:
Sorry for my late response.

I updated your code to work with the newest version (in this repo, PyPI is actually slightly outdated), and added some comments about what I changed:

```py
from pylabrobot.liquid_handling import LiquidHandler
from pylabrobot.liquid_handling.backends.opentrons_backend import OpentronsBackend 
from pylabrobot.liquid_handling.backends.simulation.simulator_backend import SimulatorBackend
from pylabrobot.resources import Coordinate
from pylabrobot.resources.opentrons import OTDeck # pylabrobot.liquid_handling.resources is now pylabrobot.resources

from pylabrobot.resources.opentrons import (
    opentrons_96_tiprack_20ul    
)
from pylabrobot.resources.corning_costar.plates import  (
    Cos_96_Rd
)

host_ip = ''
def prep_deck(device):
    Source_plate_type = Cos_96_Rd
    Destination_plate_type = Cos_96_Rd
    
    tip_rack = opentrons_96_tiprack_20ul(name="tip_rack")
    source_plate = Source_plate_type(name="source_plate")
    destination_plate = Destination_plate_type(name="destination_plate")
    
    device.deck.assign_child_at_slot(tip_rack, slot=1)
    device.deck.assign_child_at_slot(source_plate, slot=2)
    device.deck.assign_child_at_slot(destination_plate, slot=3)

async def offset_explore(device): # this is now an async function
    tip_rack: TipRack = device.get_resource("tip_rack")
    source_plate: Plate = device.get_resource("source_plate")
    destination_plate: Plate = device.get_resource("destination_plate")

    tip_source = "A1"
    await device.pick_up_tips(tip_rack[tip_source]) # added await to the device methods
    for i in range(0,10):
        await device.aspirate(
            source_plate["A1"],
            vols=[1],
            offsets=Coordinate(x=0, y=0, z=-1*i/5)) # offset now takes a Coordinate
    await device.dispense(source_plate["A1"], vols=[10]) # dispense excess liquid before putting tip back
    await device.drop_tips(tip_rack[tip_source]) # discard_tip now means put in trash, drop_tip is the new API

backend = OpentronsBackend(host=host_ip, port=31950) 
simulator = SimulatorBackend() 

lh = LiquidHandler(backend=backend, deck=OTDeck())
sim = LiquidHandler(backend=simulator, deck=OTDeck())

await sim.setup()
simulator.wait_for_connection() # wait for the browser to connect
prep_deck(sim)

# disable PLR volume trakcing
from pylabrobot.resources import set_volume_tracking
set_volume_tracking(enabled=False)

# update the simulator state with tips and volume
await simulator.fill_tip_rack(sim.deck.get_resource("tip_rack"))
await simulator.adjust_well_volume(sim.deck.get_resource("source_plate"), pattern=[[400]*12]*8)

await offset_explore(sim)
```

This code runs fine on the simulator. Could you confirm that it works for you too?

Unfortunately we lost access to our OT, so I have not been able to test on device, but it sounds like your issue has to do with the simulator. If you encounter any problems on the OT, please feel free to open a new issue.

JamesBagley, 2023-02-27:
This is great, thank you! I'll update to the newest version and try to run it on the simulator and on the OT2 when I get the chance

rickwierenga, 2023-09-21:
Did you have any luck getting this to work?

JamesBagley, 2023-09-29:
Hey Rick sorry had to drop this for a while but am taking a look at it again, do you know if the code you shared should work on the version of pylabrobot installed by pip? it seems to cause issues in the imports

rickwierenga, 2023-09-29:
The version on PyPI is definitely not up to date. I recommend installing from source instead. (https://docs.pylabrobot.org/installation.html)

closed 2024-02-26 by rickwierenga