PyLabRobot/pylabrobot #26: Picking up 2 tips on a 1-channel chatterbox backend fails with unclear error

naikymen, 2023-04-03:
I understand that this is not the intended usage, but naive users like me would (perhaps) land in this spot too, and get stuck.

[image]

It seems that PLR is not "realising" that the backend has 1 channel, and failing to use the second channel (with "Key 1").

I'd like to suggest adding a check to (at least) the `pick_up_tips` command, that handles this kind of inconsistency.

I'd suggest a spot but I'm have no clue where (yet). Let me know if I can help.

naikymen, 2023-04-03:
A similar error appears with `aspirate`:

[image]

> Note that as a regular user of the R language, I tend to assume that all operations are vectorized. I expected `lh.aspirate(plate["A1:A2"], vols=[100.0, 50.0])` to mean "aspirate 100 from A1 into channel 1, and then 50 from A2 into channel 1."

rickwierenga, 2023-04-03:
Thanks for reporting!

Fixed with https://github.com/PyLabRobot/pylabrobot/commit/48fb5cda5c888fea0f7257e73881974b829bbdd6.

rickwierenga, 2023-04-03:
> Note that as a regular user of the R language, I tend to assume that all operations are vectorized. I expected lh.aspirate(plate["A1:A2"], vols=[100.0, 50.0]) to mean "aspirate 100 from A1 into channel 1, and then 50 from A2 into channel 1."

Correct

closed 2023-04-03 by rickwierenga

naikymen, 2023-04-04:
Thank you Rick :) I'll be making some basic hardware tests this weekend on
a rudimentary backend.

Best!

naikymen, 2024-05-10:
Hi Rick, sorry to bring this up again.

> > Note that as a regular user of the R language, I tend to assume that all operations are vectorized. I expected lh.aspirate(plate["A1:A2"], vols=[100.0, 50.0]) to mean "aspirate 100 from A1 into channel 1, and then 50 from A2 into channel 1."
> 
> Correct

I just want to check if that really is correct.

I am still getting the error:

[image]

I think the vectorization in those functions is only channel-wise (i.e. to operate multi-channel pipettes).

rickwierenga, 2024-05-10:
> aspirate 100 from A1 into channel 1, and then 50 from A2 into channel 1."
Channels are actually 0-indexed. Excuse the oversight: 100 from A1 into channel `0` :)

The error you see is because you're aspirating from two wells simultaneously (A1 and A2), whereas your backend `echo` only has one channel (as set by the `num_channels` property).

All atomic operations (`aspirate`, `pick_up_tips`, `drop_tips`, `dispense`) are expected to be executed simultaneously. Big robots like STAR support this up to 16 channels, although 8 is typical. If the user wants to do things separately, they should make two method calls on `lh`.

> sorry to bring this up again.

this is technically a separate issue, but in general please don't hesitate to tell me about anything you think can be improved.