PyLabRobot/pylabrobot #31: Issues in the simulator notebook

tianar, 2023-08-01:
1. `await sb.adjust_well_volume(plate_1, pattern=[[500]*12]*8)` cell never finishes execution. I removed the `await` to proceed with the notebook.

2.
```
plate_1.set_well_volumes([[500]*12]*8)
plate_2.set_well_volumes([[100, 500]*6]*8)
```
gives an error as the method `set_well_volumes` was replaced by `set_well_liquids` in commit [9faac49](https://github.com/PyLabRobot/pylabrobot/commit/9faac49571b5eec111853dab5cfd895429fd97da) on June 19.
My workaround for this was:
```
from pylabrobot.resources.liquid import Liquid
plate_1.set_well_liquids((Liquid.WATER, [[500]*12]*8))
plate_2.set_well_liquids((Liquid.WATER, [[100, 500]*6]*8))
```
However, nothing changes in the simulator, volumes are not assigned:
[image]

3. If I continue executing other cells I get stuck at `aspirate` - the `await lh.aspirate(plate["A2"], vols=[300])` cell never gets executed.
The same happens with `await lh.aspirate_plate(plt_car[0].resource, volume=200)`

Note: PyLabRobot was installed from the source.

Thank you very much for working on this!

rickwierenga, 2023-08-01:
Thank you for opening this issue!

I'm aware that the simulator is currently broken (due to me refactoring things for the GUI). I will fix it soon (hopefully before the end of the week) and ping you when I'm done. 

In the meantime, if urgent, I would recommend using an older version of the library (before https://github.com/PyLabRobot/pylabrobot/commit/a54acb3385e89b7e0ec189068e5f36593bd1077c).

tianar, 2023-08-01:
Thanks for your reply and suggestion! I tried an older version #[38978e7](https://github.com/PyLabRobot/pylabrobot/commit/38978e71e9a14e125e7debd402da33f0fb467e68)  but I'm getting an import error.

```
ImportError: cannot import name 'LiquidClass' from 'pylabrobot.liquid_handling.liquid_classes' 
```

I think the best is to wait for your fix. Many thanks!

rickwierenga, 2023-08-02:
https://github.com/PyLabRobot/pylabrobot/commit/861d2e9c3e205323bf7df89823a086662b775545 should address this. Could you confirm that the notebook is working on your end?

tianar, 2023-08-02:
Thank you for promptly addressing this issue, I apreciate it!

Unfortunately, I'm getting an error after running `await sb.adjust_wells_liquids(plate_1, liquids=plate_1_liquids)`:
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[26], line 2
      1 plate_1_liquids = [[(None, 500)]]*96
----> 2 await sb.adjust_wells_liquids(plate_1, liquids=plate_1_liquids)

File ~/Documents/Code/pylabrobot/pylabrobot/liquid_handling/backends/simulation/simulator_backend.py:207, in SimulatorBackend.adjust_wells_liquids(self, plate, liquids)
    197 for well, well_liquids in zip(plate.get_all_items(), liquids):
    198   serialized_pattern.append({
    199     "well_name": well.name,
    200     "liquids": [
   (...)
    204       } for liquid, volume in well_liquids]
    205   })
--> 207 await self.send_command(command="adjust_well_liquids", data={"pattern": serialized_pattern})

File ~/Documents/Code/pylabrobot/pylabrobot/liquid_handling/backends/websocket.py:223, in WebSocketBackend.send_command(self, command, data, wait_for_response)
    221     if not message["success"]:
    222       error = message.get("error", "unknown error")
--> 223       raise RuntimeError(f"Error during event {command}: " + error)
    225     return message
    227 return None

RuntimeError: Error during event adjust_well_liquids: Unknown event
```

rickwierenga, 2023-08-02:
I suspect this is due to caching in your browser. Could you force reload the page? (shift command r for me in Chrome/Safari)

tianar, 2023-08-02:
It worked! Thank you so much!

Before you close this issue, I wanted to share another problem I had with the simulator notebook.

After importing `from pylabrobot.liquid_handling import LiquidHandler` I get:
```
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
Cell In[3], line 1
----> 1 from pylabrobot.liquid_handling import LiquidHandler
      2 from pylabrobot.liquid_handling.backends.simulation.simulator_backend import SimulatorBackend

File ~/Documents/Code/pylabrobot/pylabrobot/liquid_handling/__init__.py:3
      1 """ Liquid handling module for PyLabRobot """
----> 3 from .backends import STAR, SerializingSavingBackend
      4 from .liquid_handler import LiquidHandler
      5 from .standard import (
      6   Pickup,
      7   Drop,
   (...)
     14   Move
     15 )

File ~/Documents/Code/pylabrobot/pylabrobot/liquid_handling/backends/__init__.py:5
      3 from .backend import LiquidHandlerBackend
      4 from .serializing_backend import SerializingBackend, SerializingSavingBackend # many rely on this
----> 5 from .websocket import WebSocketBackend # simulation relies on websocket backend
      7 from .USBBackend import USBBackend
      8 from .hamilton.errors import HamiltonError

File ~/Documents/Code/pylabrobot/pylabrobot/liquid_handling/backends/websocket.py:29
     23   import websockets.legacy
     26 logger = logging.getLogger(__name__) # TODO: get from somewhere else?
---> 29 class WebSocketBackend(SerializingBackend):
     30   """ A backend that hosts a websocket server and sends commands over it. """
     32   def __init__(
     33     self,
     34     num_channels: int,
     35     ws_host: str = "127.0.0.1",
     36     ws_port: int = 2121,
     37   ):

File ~/Documents/Code/pylabrobot/pylabrobot/liquid_handling/backends/websocket.py:114, in WebSocketBackend()
    111   if event == "ping":
    112     await self.websocket.send(json.dumps({"event": "pong"}))
--> 114 async def _socket_handler(self, websocket: websockets.legacy.server.WebSocketServerProtocol):
    115   """ Handle a new websocket connection. Save the websocket connection store received
    116   messages in `self.received`. """
    118   while True:

NameError: name 'websockets' is not defined
```
which is odd, as `websockets` is indeed installed. My workaround was:
```
import sys
sys.path.append('../env/lib/python3.10/site-packages')
import websockets
```
before importing `pylabrobot`.

Thank you again for fixing the notebook and the amazing work on this package!

rickwierenga, 2023-08-02:
Hm, that's interesting. Do you experience this issue with any other packages?

tianar, 2023-08-02:
Nope, this is the only one so far.

rickwierenga, 2023-08-03:
Be sure to use the python version from your virtual environment if you want to use the libraries installed there. Please file a new issue if this error occurs with other libraries and you're sure you're using the right python installation.

closed 2023-08-03 by rickwierenga