PyLabRobot/pylabrobot #33: Error in Tecan tip rack functions

tianar, 2023-08-04:
Only the `DiTi_1000ul_LiHa` and `DiTi_10ul_LiHa` tip racks have the required `make_tip` argument. Others result in error:
```
TypeError: TipSpot.__init__() missing 1 required positional argument: 'make_tip'
```

tianar, 2023-08-04:
The next issue is with the `pick_up_tips()` function.

I've modified the simulation tutorial to work with Tecan deck `EVO150Deck` instead of Hamilton `STARLetDeck`,  tip carrier `TIP_CAR_480_A00`->`DiTi_SBS_4_Pos_MCA96`  and tip rack `HTF_L`-> `DiTi_1000ul_LiHa`.
Unfortunately, `await lh.pick_up_tips(tip_0["A1", "B2"])` never gets executed.

rickwierenga, 2023-12-27:
Added tip creators for Tecan DiTi with https://github.com/PyLabRobot/pylabrobot/commit/f1873027187531a3f59a91393de5fbf115c2fd95.

The simulator has changed a lot since the summer. Please open a new issue if the you're still having trouble with it.

closed 2023-12-27 by rickwierenga