PyLabRobot/pylabrobot #60: Simulator not updating

BioCam, 2024-01-15:
Since at least yesterday but not four days ago, I noticed that the PLR simulator is not updating.

Minimal code to reproduce issue:
```
%load_ext autoreload
%autoreload 2

from pylabrobot.liquid_handling import LiquidHandler
from pylabrobot.resources.hamilton import STARLetDeck
from pylabrobot.liquid_handling.backends.simulation.simulator_backend import SimulatorBackend

backend = SimulatorBackend(open_browser=True)
lh = LiquidHandler(backend=backend, deck=STARLetDeck())

await lh.setup()

from pylabrobot.resources import (
    TIP_CAR_480_A00, # TIP CARRIERS
    TIP_50ul_L # TIP Racks
)

# Generate TIP carrier instances
tip_car01 = TIP_CAR_480_A00(name='tip_carrier_1')

# Generate tip rack instances
lh.deck.assign_child_resource(tip_car01, rails=1)
tip_car01[0] = TIP_50ul_L(name='tips_00_50ul')
# Retrieve tip rack information
tip_00_50ul = lh.get_resource("tips_00_50ul")
```

Yet, after running this script the simulator is still just the blank deck:
[image]

Furthermore,
```
await backend.fill_tip_rack(tip_00_50ul)
```
...raises the ValueError:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[5], line 1
----> 1 await backend.fill_tip_rack(tip_00_50ul)

File /mnt/c/Users/cmoschner/pylabrobot/pylabrobot/liquid_handling/backends/simulation/simulator_backend.py:267, in SimulatorBackend.fill_tip_rack(self, resource)
    259 async def fill_tip_rack(self, resource: TipRack):
    260   """ Completely fill a :class:`~pylabrobot.resources.TipRack` resource
    261   with tips. (**simulator only**).
    262 
    263   Args:
    264     resource: The resource where all tips should be placed.
    265   """
--> 267   await self.edit_tips(resource, [[True] * 12] * 8)

File /mnt/c/Users/cmoschner/pylabrobot/pylabrobot/liquid_handling/backends/simulation/simulator_backend.py:257, in SimulatorBackend.edit_tips(self, tip_rack, pattern)
    251     tip = tip_rack.get_item(idx)
    252     serialized_pattern.append({
    253       "tip": tip.serialize(),
    254       "has_one": has_one
    255     })
--> 257 await self.send_command(command="edit_tips", data={"pattern": serialized_pattern})

File /mnt/c/Users/cmoschner/pylabrobot/pylabrobot/liquid_handling/backends/websocket.py:208, in WebSocketBackend.send_command(self, command, data, wait_for_response)
    206 # Run and save if the websocket connection has been established, otherwise just save.
    207 if wait_for_response and not self.has_connection():
--> 208   raise ValueError("Cannot wait for response when no websocket connection is established.")
    210 if self.has_connection():
    211   asyncio.run_coroutine_threadsafe(self.websocket.send(serialized_data), self.loop)

ValueError: Cannot wait for response when no websocket connection is established.
```

But the simulator in the image above indicates that a connection has been established and opened in a separate window.

rickwierenga, 2024-01-15:
My initial guess is that there is another simulator / process running on the same port, causing the simulator server to run on another port. When you open the simulator, it would have connected to the wrong backend, essentially. Could you post the output of `setup`?

If you can't find the process and stop it, you can use `kill $(lsof -t -i:2121)` (and 1337) on linux/mac to kill it.

BioCam, 2024-01-15:
Thank you, Rick.

I killed every possible running process by simply restarting the Windows and Mac machines on which I reproduced the issue and it is working again.

rickwierenga, 2024-01-15:
Great.

I'd like to keep this issue open because I have to adjust the frontend to automatically connect to the right backend.

closed 2024-01-22 by rickwierenga