PyLabRobot/pylabrobot #79: Is it possible to use Tecan Fluent as backend?

PraljakReps, 2024-03-18:
I enjoyed reading the peer-reviewer paper on pylabrobot! Question: I noticed that the wrapper is OS-agnostic and even robot-agnostic. It can even handle Tecan Evo as the backend. However, is it possible to use pylabrobot for **Tecan Fluent** instead?

rickwierenga, 2024-03-18:
Glad you like the project! Unfortunately, the Fluent is not supported yet. Are you interested in developing this?

PraljakReps, 2024-03-18:
I can try to extend the project to Tecan Fluent. Do you have any initial pointers or tips when working with Tecan software as the backend?

rickwierenga, 2024-03-18:
Awesome!

I assume you have a computer with the Tecan software installed. This program generates log files of commands sent to the machine, which PyLabRobot should be made to generate and send instead. These logs are located in C:/programdata/tecan/evoware/audittrail/log for the evo, and it's likely something similar for fluent. From there, we'll need to figure out how exactly the Tecan software communicates with the machine. I'll look at the current EVO backend and see if there are any obvious similarities we can exploit. If you could share the log files you find, I'll be able to explain things in more detail.

Can you me at rick_wierenga@icloud.com?

stefangolas, 2024-04-10:
@PraljakReps we actually have collaborators interested in doing a Fluent integration at the University of Tennessee, I can also put you in touch with them

rickwierenga, 2024-06-05:
Closing this issue. @PraljakReps please get in touch to discuss further if you're still interested :)

closed 2024-06-05 by rickwierenga

clangi, 2024-06-11:
I am also interested in the possibility to use it on a Tecan Fluent as back-end. Are you opening a new dev branch for this purpose?

rickwierenga, 2024-06-11:
Great!

Sure, we can do a dev branch when someone actually starts working on it. Currently, I don't think anyone has this in progress.