PyLabRobot/pylabrobot #123: TooLittleLiquidError raised on dispense with volume tracking on notebook example

naikymen, 2024-05-06:
Hi Rick, how are you doing?

I got to work and may have found an issue with volume trackers during a dispense operation. It seems that volumes and liquids in the target wells are messed up.

This happens only when installing PLR from the repo. Installing from pypi does not show the issue. It took me a while to realise that it may not have been me messing up something.

Here is my evidence:

- https://colab.research.google.com/drive/1CfALtxgMvBBB-osH9ofVBBZf4QUwOhwW?usp=sharing
  - Print of the above: [PLR-test.ipynb - Colab.pdf](https://github.com/PyLabRobot/pylabrobot/files/15217536/PLR-test.ipynb.-.Colab.pdf)

Best!
Nico

PS: I'm working on generating a deck from my data, and trying to implement "tube racks" savagely. I thought I'd use the latest commit for this. Is there another commit I should be using for work? Perhaps one tied to the pypi package?

naikymen, 2024-05-06:
For a quick view of a symptom:

[image]

naikymen, 2024-05-07:
I think I tracked down the issue to e77d259e1785e9c4b36187fe1ae71022c3d7343d

Here is a script for quick testing: 

<details>

<summary>test script</summary>

```python
#!/usr/bin/env python

from pylabrobot.liquid_handling import LiquidHandler

from pylabrobot.liquid_handling.backends.chatterbox_backend import ChatterBoxBackend

from pylabrobot.resources.hamilton import STARLetDeck

from pylabrobot.resources import set_tip_tracking, set_volume_tracking

async def test():

    set_volume_tracking(enabled=True)
    set_tip_tracking(enabled=True)

    backend = ChatterBoxBackend()

    lh = LiquidHandler(backend=backend, deck=STARLetDeck())

    await lh.setup()

    from pylabrobot.resources import (
        TIP_CAR_480_A00,
        PLT_CAR_L5AC_A00,
        Cos_96_DW_1mL,
        HTF_L
    )

    tip_car = TIP_CAR_480_A00(name='tip carrier')
    tip_car[0] = HTF_L(name='tips_01')

    lh.deck.assign_child_resource(tip_car, rails=3)

    plt_car = PLT_CAR_L5AC_A00(name='plate carrier')
    plt_car[0] = Cos_96_DW_1mL(name='plate_01')

    lh.deck.assign_child_resource(plt_car, rails=15)

    from pylabrobot.resources.liquid import Liquid

    plate = lh.get_resource("plate_01")
    plate.set_well_liquids((Liquid.WATER, 300))

    lh.summary()

    tiprack = lh.get_resource("tips_01")
    await lh.pick_up_tips(tiprack["A1:C1"])

    plate = lh.get_resource("plate_01")

    well = plate.get_item("A1")
    print(well.serialize_all_state())
    print(well.tracker.get_used_volume())

    await lh.aspirate(plate["A1:C1"], vols=[100.0, 50.0, 200.0])

    well = plate.get_item("A1")
    print(well.serialize_all_state())
    print(well.tracker.get_used_volume())

    well = plate.get_item("D1")
    print(well.serialize_all_state())
    print(well.tracker.get_used_volume())

    await lh.dispense(plate["D1:F1"], vols=[100.0, 50.0, 200.0])

    well = plate.get_item("D1")
    print(well.serialize_all_state())
    print(well.tracker.get_used_volume())

    await lh.aspirate(plate["D1:F1"], vols=[100.0, 50.0, 200.0])
    await lh.dispense(plate["A1:C1"], vols=[100.0, 50.0, 200.0])

    well = plate.get_item("A1")
    print(well.serialize_all_state())
    print(well.tracker.get_used_volume())
    well = plate.get_item("D1")
    print(well.serialize_all_state())
    print(well.tracker.get_used_volume())

    await lh.drop_tips(tiprack["A1:C1"])

    await lh.stop()

import asyncio

asyncio.run(test())
```

</details>

rickwierenga, 2024-05-07:
doing well! sorry for the late reply. thanks for the test script.

fixed with https://github.com/PyLabRobot/pylabrobot/commit/c655ceb5b6d5a4fbece10b54420ec0562728a82c.

closed 2024-05-07 by rickwierenga

naikymen, 2024-05-07:
Sweet, thanks Rick :)