PyLabRobot/pylabrobot #125: Backend for Biomek i5/i7

Shaun-Regenbaum, 2024-05-08:
I was wondering if anyone has worked on integrating beckman coulter machines with this library? Or if not, someone who I could perhaps talk to to contribute some new backends?

rickwierenga, 2024-05-08:
We don't have any BC machines yet and as far as I know, none of the current contributors have one.

Do you have one / would you be interested in contributing this?

Shaun-Regenbaum, 2024-05-22:
We do have one, and am interested in contributing, not sure how much work
will be involved yet, so would love to get an idea.

rickwierenga, 2024-05-23:
great! do you have firmware documentation? / have you used wireshark before?

if you want, i'd be happy to go on a call and explain, or we can just continue in this thread 🤷‍♂️

Shaun-Regenbaum, 2024-05-23:
A call would be great, perhaps next week?
I have some experience in related fields, but no it would be a new thing
for me.

rickwierenga, 2024-05-23:
sure! please email me at wierenga at mit.edu so we can plan something :)

dejmail, 2025-04-30:
Was wondering if any progress had been made on this?