PyLabRobot/pylabrobot #170: Remove Resource's that have catalogue number or other identifier

rickwierenga, 2024-06-24:
It appears many resources imported from the VENUS labware library have no catalogue number, and are therefore not usable or reliable. These will need to be removed in the future. We will need to rely on a combination of reliable third party databases, information from manufacturers, and crowd-sourced definitions.

As per @BioCam's [definition](https://github.com/PyLabRobot/pylabrobot/pull/169), going forward all resources will need "PLR-definition of the physical object + metadata needed to use it". Camillo has documented all resources that qualify in the associated README files in the `pylabrobot.resources` submodule, complete with identifiers and pictures.

I will be adding exceptions to resource definitions which I believe no one has verified. This will give people the chance to communicate that the definitions _are_ correct, or what changes are needed.

Resources for which we have not heard anything will be removed on September 1st.

BioCam, 2024-06-24:
Sorry, @rickwierenga, did you mean "Remove resources that **do not** have a catalogue number or other identifier"?

rickwierenga, 2024-06-27:
smh

jonlaurent, 2024-08-30:
Hey guys, I am starting to boot up some PLR stuff again and catching up on some big changes apparently. It seems like this particuarly change is going to remove quite a lot of plates in a few days, no? Like just about every Costar plate, for example.

I can help get some new plates in. I've previously been using some random Costar or other plate defs that happen to work for non-Costar plates, so pretty much everything is broken on this lol.

In particular: 384 PCR plates (from Globe, but ABI or Roche copies), Echo plates (from Labcyte or Beckman), Costar deep wells (which I think I can just change to Cos_384_wellplate_2mL... for now) and some plate reader plates I'll have to look at again.

jonlaurent, 2024-08-30:
Although looks like Costar stuff may be coming via this: [https://github.com/PyLabRobot/pylabrobot/issues/171](https://github.com/PyLabRobot/pylabrobot/issues/171) ??

rickwierenga, 2024-08-30:
Sure yeah please create any plate definitions for the stuff that you're using!

#171 is a handy data sheet, but not a soon-to-do for me (is generally good to verify defs, and I don't have many of the plates listed there). use it as a starting point for your labware :)

closed 2024-10-22 by rickwierenga