PyLabRobot/pylabrobot #176: Create a distribute feature

rickwierenga, 2024-06-26:
like https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.InstrumentContext.distribute

Can lift from https://github.com/rickwierenga/pylabrobot-art-studio, where this is already implemented. Just need to make it nice and general

rickwierenga, 2024-07-01:
@BioCam is the existing `transfer` sufficient for your needs?

BioCam, 2024-07-02:
I am just reading through the `liquid_handler.transfer()` function. It is quite a neat, small function:

```python
  async def transfer(
    self,
    source: Well,
    targets: List[Well],
    source_vol: Optional[float] = None,
    ratios: Optional[List[float]] = None,
    target_vols: Optional[List[float]] = None,
    aspiration_flow_rate: Optional[float] = None,
    dispense_flow_rates: Optional[Union[float, List[Optional[float]]]] = None,
    **backend_kwargs
  ):
    """Transfer liquid from one well to another.

    Examples:

      Transfer 50 uL of liquid from the first well to the second well:

      >>> lh.transfer(plate["A1"], plate["B1"], source_vol=50)

      Transfer 80 uL of liquid from the first well equally to the first column:

      >>> lh.transfer(plate["A1"], plate["A1:H1"], source_vol=80)

      Transfer 60 uL of liquid from the first well in a 1:2 ratio to 2 other wells:

      >>> lh.transfer(plate["A1"], plate["B1:C1"], source_vol=60, ratios=[2, 1])

      Transfer arbitrary volumes to the first column:

      >>> lh.transfer(plate["A1"], plate["A1:H1"], target_vols=[3, 1, 4, 1, 5, 9, 6, 2])

    Args:
      source: The source well.
      targets: The target wells.
      source_vol: The volume to transfer from the source well.
      ratios: The ratios to use when transferring liquid to the target wells. If not specified, then
        the volumes will be distributed equally.
      target_vols: The volumes to transfer to the target wells. If specified, `source_vols` and
        `ratios` must be `None`.
      aspiration_flow_rate: The flow rate to use when aspirating, in ul/s. If `None`, the backend
        default will be used.
      dispense_flow_rates: The flow rates to use when dispensing, in ul/s. If `None`, the backend
        default will be used. Either a single flow rate for all channels, or a list of flow rates,
        one for each target well.

    Raises:
      RuntimeError: If the setup has not been run. See :meth:`~LiquidHandler.setup`.
    """

    # Deprecation check for single values
    if isinstance(targets, Well):
      raise NotImplementedError("Single target is deprecated, use a list of targets.")
    if isinstance(dispense_flow_rates, numbers.Rational):
      raise NotImplementedError("Single dispense flow rate is deprecated, use a list of flow rates")

    if target_vols is not None:
      if ratios is not None:
        raise TypeError("Cannot specify ratios and target_vols at the same time")
      if source_vol is not None:
        raise TypeError("Cannot specify source_vol and target_vols at the same time")
    else:
      if source_vol is None:
        raise TypeError("Must specify either source_vol or target_vols")

      if ratios is None:
        ratios = [1] * len(targets)

[... 38 lines cut ...]
Instead of giving the function a transfer volume list a **transfer volume matrix** is, as the name implies, a 3-dimensional array: 0-7 rows representing each channel (in an 8-channel system, more if more channels exist), but it has no real limit in the number columns.
 
A simple showcase to highlight the power I imagine this to have:

We all perform normalizations, no secret here. Let's say you have a 96-well plate of protein extractions one needs to normalise.
We calculate that we need to add the following buffer to a new well, and then add x volume from the extraction plate to it:

[image]

There is no reason to transfer all these volumes of the same buffer to an empty well in separate transfers, `distributing` / `aliquotting` is way faster, particularly if when one scales to multiple plates.
But at the same time using a single tip (like withe OT-2) is painfully slow.

This is where PLR's new `distribute` function can come in: use all channels available on the robot one has; the function is simply given the matrix of destination resources, and a matrix of volumes, and then identifies how to chunk the matrix - column by column - into aspiration cycles to execute.

It would be vastly faster than anything currently available and totally in the realm of y-independent channel systems ... the only thing that's missing is a flexible-enough, detailed-enough, computational design-empowered software = PLR :)

Normalisation is just one application, I can see use cases like inoculations, experimental setup construction and more.

This function should also have arguments like `pre_aspirate_dispense=[5,5,5,None]` (i.e. the volume to dispense straight back to the source well to avoid the known issue with the first aliquot), and  `retention_vols=[5,5,5,None]` (i.e. the volume that should stay back in the tip to ensure the last aliquot is identical to all the previous ones).

This is a big vision for a PLR method imo but I am proposing it because I believe it to be incredibly powerful, and a way to set PLR instantly apart and showcase what programmationally-designed liquid handling can do.

---

To be clear, I don't believe the information regarding channel parameter settings to make such a function accurate and reliable is given with the current clinging to "liquid classes", but I'm sure we can figure out a better way using the full set of aspiration and dispensation arguments ;)

rickwierenga, 2024-07-02:
I will make this

BioCam, 2024-07-02:
hahaha 🤣 

I thought you might like it

BioCam, 2024-07-02:
> I will make this

I know how much fun it will be to actually build this function, so I won't take this from you @rickwierenga (plus, I've got a ton of work in my day job and other PLR dev tasks) but how about we have a meeting solely discussing how to implement this function in the next couple of day?

For example, we must absolutely polish `STAR.aspirate()` and `STAR.dispense()` before heading into this (unit standardisation in particular) and we must ensure we have access to all arguments. I see these tasks as urgent precedence constraints to the implementation of `lh.distribute()` / `lh.aliquot()`.

PS. I did warn you that once PLR is more open to complex, more "user-focused" features I will let a lot of ideas loose :P

rickwierenga, 2024-07-02:
> a meeting

Sure, what time?

> PS. I did warn you that once PLR is more open to complex, more "user-focused" features I will let a lot of ideas loose :P

This has always been the goal: front ends exist to compose complex functions out of easy-to-implement atomic commands. (obviously while maintaining granular control.) From the paper: "Composition of the unit operations, such as the discard operation (drop tip in trash) and the transfer operations (combined aspirate and dispense), are performed at the LiquidHandler level and above". So excited we are nearing a stage where we can productively implement composite commands more than simple `transfer` and `discard` 🎉

BioCam, 2024-07-06:
@rickwierenga, I'm available today until 15:00 UCT+1 or on Wednesday? :)

rickwierenga, 2024-07-06:
sorry to delay, but Wednesday is a lot better for me

BioCam, 2024-07-06:
I know, let's do Wednesday during the day at 11:00 UCT+1?
This meeting shouldn't be longer than 45 min because we just want to create a requirements list for the distribute function to begin the initial design; there are quite a lot of design considerations tbh.

rickwierenga, 2024-07-06:
Perfect!