PyLabRobot/pylabrobot #197: OT flex support

ashah03, 2024-07-17:
Any plans to support the Opentrons Flex robot?

rickwierenga, 2024-07-17:
Sure, but we don't currently have anyone with this robot. Do you have one? Do you want to work on this?

ashah03, 2024-07-17:
Not yet, but I may get one / be able to help in the future

rickwierenga, 2024-07-17:
Grea! Let me know!

rickwierenga, 2024-07-17:
closing this for now

closed 2024-07-17 by rickwierenga

dabdoue, 2025-02-24:
I have a Flex and am interested in this software. I have a pretty quick turnaround for a project where I want to use the Flex along with a 6 axis robot in a ROS2 environment, and having access to software like this would be fantastic for future modularity.

Please feel free to reach out to me on my LinkedIn (GH profile) so we can establish more formal and personal communication

rickwierenga, 2025-02-25:
sounds cool! would be down to write backend for the flex and/or robot arm?