PyLabRobot/pylabrobot #214: ModuleNotFoundError upon installation

lennijusten, 2024-08-09:
When installing `pylabrobot==0.1.6` and running
```
from pylabrobot.liquid_handling import LiquidHandler
```
I get an error:
```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[1], [line 1](vscode-notebook-cell:?execution_count=1&line=1)
----> [1](vscode-notebook-cell:?execution_count=1&line=1) from pylabrobot.liquid_handling import LiquidHandler
      [2](vscode-notebook-cell:?execution_count=1&line=2) from pylabrobot.liquid_handling.backends import ChatterBoxBackend
      [3](vscode-notebook-cell:?execution_count=1&line=3) from pylabrobot.visualizer.visualizer import Visualizer

File ~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:2
      [1](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:1) from .backends import *
----> [2](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:2) from .liquid_handler import LiquidHandler
      [3](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:3) from .standard import (
      [4](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:4)   Pickup,
      [5](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:5)   Drop,
   (...)
     [12](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:12)   Move
     [13](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:13) )
     [14](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/__init__.py:14) from .strictness import Strictness, set_strictness, get_strictness

File ~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:44
     [21](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:21) from pylabrobot.resources import (
     [22](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:22)   Container,
     [23](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:23)   Deck,
   (...)
     [41](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:41)   does_cross_contamination_tracking
     [42](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:42) )
...
----> [4](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/tilting/hamilton_backend.py:4) import serial
      [6](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/tilting/hamilton_backend.py:6) from pylabrobot.tilting.tilter_backend import TilterBackend, TiltModuleError
      [9](https://file+.vscode-resource.vscode-cdn.net/Users/lenni/Documents/GitHub/intelligent-lab/notebooks/pylabrobot/~/Documents/GitHub/intelligent-lab/pylabrobot/pylabrobot/tilting/hamilton_backend.py:9) class HamiltonTiltModuleBackend(TilterBackend):

ModuleNotFoundError: No module named 'serial'
```

rickwierenga, 2024-08-09:
is this installing from PyPI? The PyPI version is (very) outdated, please install from source instead

rickwierenga, 2024-08-09:
actually, also an error on plr:main. will fix now

rickwierenga, 2024-08-09:
https://github.com/PyLabRobot/pylabrobot/commit/08a36664fdbac871336d5b2ada2952678af0ce2b should fix it

lennijusten, 2024-08-09:
Yup, thanks. I'm using a version cloned from source.

closed 2024-08-09 by lennijusten

rickwierenga, 2024-08-09:
thanks for reporting!