PyLabRobot/pylabrobot #225: remove Hamilton liquid classes

rickwierenga, 2024-08-22:
Liquid classes in the Hamiltonian sense are confusing, especially in PLR: https://labautomation.io/t/how-does-pylabrobot-handle-liquid-classes/977

A lot of code in STAR looks like this:

```python
flow_rates = [
      op.flow_rate or (hlc.aspiration_flow_rate if hlc is not None else 100)
        for op, hlc in zip(ops, hamilton_liquid_classes)]
```

Hamiltonian liquid classes are effectively dicts of parameters.

We should do the following instead:

```python
lh.aspirate(..., **hamilton_liquid_class.generate_kwargs())`
```

```python
>>> hamilton_liquid_class.generate_kwargs()
{
  "flow_rate": 42,
  "clot_detection_height": 69,
  ...
}
```

so that we can remove all references to Hamilton liquid classes in `STAR`/`Vantage` backends, thus simplifying them a lot. `HamiltonLiquidClass` can simply be a [TypedDict](https://peps.python.org/pep-0589/).

(I vaguely recall discussing this before, possibly in a private message, can't find a public issue/forum post now)

rickwierenga, 2024-09-11:
one issue with making it a TypedDict is that volume correction becomes slightly more awkward, and the user needs to explicitly say

```python
lh.aspirate(vols=[hlc.correct(original_volume)])
```

but i think for sensitive liquid handling you'll want to fine-tine the volumes anyway, and this method is a lot more transparent than doing it somewhere deep in STAR where it's hidden

rickwierenga, 2024-09-18:
first PR: https://github.com/PyLabRobot/pylabrobot/pull/248

BioCam, 2024-09-27:
> one issue with making it a TypedDict is that volume correction becomes slightly more awkward, and the user needs to explicitly say
> 
> ```python
> lh.aspirate(vols=[hlc.correct(original_volume)])
> ```

I agree; this puts enormous technical debt on the end user.

This way every user must constantly think about the actual piston_volume, i.e. the "corrected_volume", one has to move to achieve the target_liquid_volume.

The excellent idea behind standard liquid classes is to not have worry about this.
It makes reading and writing complex protocols much harder.

Ideally we want a system/advanced_liquid_class that is **more** than a dictionary, i.e. an object that corrects not just for volume but corrects all aspirate/dispense attributes... because why would the same blow_out_volume for 1 ul of 50% glycerol work the same way for 1000 of 50% glycerol.

This current alternative doesn't seem to address this.

rickwierenga, 2024-09-27:
but with liquid classes, you would still have to worry about this, because the provided liquid classes aren't good enough.

yes, we don't have an 'advanced liquid class' yet.