PyLabRobot/pylabrobot #257: Plain pip install from repo fails to load LiquidHandler - Missing pylibftdi dependency

naikymen, 2024-09-22:
I'm not sure if this is a bug, but I was not expecting it.

When installing with `pip install .` from the repo's directory and running my tests, I get an import error.

The `pylibftdi` module seems to be required, even though it is part of `extras_plate_reading` in `setup.py`. I _could_ install it with `pip install .[dev]` to get going, but it does not seem optimal for the general case.

```txt
Traceback:
/usr/local/lib/python3.9/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
pylabrobot/resources/pipettin/pipettin_tests.py:4: in <module>
    from pylabrobot.liquid_handling.backends.piper_backend import PiperBackend
pylabrobot/liquid_handling/__init__.py:2: in <module>
    from .liquid_handler import LiquidHandler
pylabrobot/liquid_handling/liquid_handler.py:18: in <module>
    from pylabrobot.plate_reading import PlateReader
pylabrobot/plate_reading/__init__.py:2: in <module>
    from .biotek_backend import Cytation5Backend
pylabrobot/plate_reading/biotek_backend.py:6: in <module>
    from pylibftdi import Device
E   ModuleNotFoundError: No module named 'pylibftdi'
```

It is used in `liquid_handler.py` for a validation, and perhaps elsewhere.

[image]

I'm not sure how to fix this without adding that module in `install_requires`. I've added it for now.

Best!

rickwierenga, 2024-09-22:
thanks for flagging, made it similar to how we handle optional dependencies in other cases: https://github.com/PyLabRobot/pylabrobot/commit/f6f4693e65ccaacf1f861777ee89e6f67fef4540

closed 2024-09-22 by rickwierenga

naikymen, 2024-09-22:
Thanks Rick, you're the best!