PyLabRobot/pylabrobot #259: ItemizedResource: inconsistency in docstrings of dy, dx, dz and calculated coordinates

naikymen, 2024-09-23:
Hi! I'm confused by a behaviour here.

The docstrings state:

    dx: The bottom left corner for items in the left column
    dy: The bottom left corner for items in the top row
    dz: The z coordinate for all items

From this, I interpret that "dy" is the Y coordinate of the bottom-left corner of the items (e.g. Wells) in the top row (i.e. row A) of the ItemizedResource (e.g. A1, A2, etc.).

[image]

The `Eppendorf_96_wellplate_250ul_Vb` plate, for example, sets `dy=8.26`, which is suspiciously small.

[image]

On inspection of the Wells in column 1, `dy` ends up being the Y coordinate of the items in the **bottom** row of the ItemizedResource (i.e. H1, H2, ...), and the Well in the top row has a much larger value.

[image]

What am I getting wrong?

Best!
N

rickwierenga, 2024-09-24:
yes, you found a bug in the documentation, should be the bottom row :)

I'm currently working on big changes to the docs (https://github.com/PyLabRobot/pylabrobot/pull/249), will fix, thanks for flagging :)

rickwierenga, 2024-09-24:
https://github.com/PyLabRobot/pylabrobot/commit/6d48b34de78f0b727a200030427d9c3e176a2834 (was actually just in doc strings)

closed 2024-09-24 by rickwierenga

naikymen, 2024-09-24:
Good to know.

[image]

naikymen, 2024-09-24:
In the meantime, looking for some plate to use as a reference, I found that that Thermo plate does not match the dimensions in the datasheet exactly.

I've marked that here for now: https://gitlab.com/pipettin-bot/forks/pylabrobot/-/commit/6c470a129d87b031dd6210512c59b6721f6714d1#21ca88187b01ddf8498db898ec7fd86de595d156

Is this usually the case with the definitions of labware in PLR?

rickwierenga, 2024-09-24:
it happens that data in these datasheets does not exactly correspond with what we see in reality. i think physical reality should take precedence

(I have been encouraging people to start separate threads for better searchability :))