PyLabRobot/pylabrobot #272: [bug] Dispense operation selects the wrong channel and fails with tip and volume tracking on

naikymen, 2024-10-07:
I bring bugs!

Here is my situation, I want to:

1. Pick a tip with channel 1 ✅
2. Aspirate from a tube with channel 1  ✅
3. Dispense in 3 wells (serially) with channel 1 ❌

<details><summary>Details</summary>
<p>

Note that the "head" shows that a tip is placed in channel 1.

[image]

</p>
</details> 

I think that the wrong channel is being checked, because I get this exception message:

    NoTipError: Channel 0 does not have a tip.

Some notes:

- I selected channel 1, but PLR is looking for a tip in channel 0 (which is not in use by the dispense operation).
- It errors-out on a dispense, but not on aspirate.

Here is the traceback:

<details><summary>Details</summary>
<p>

```
---------------------------------------------------------------------------
NoTipError                                Traceback (most recent call last)
Cell In[11], line 5
      1 well_plate = deck["Standard 384-well plate"]
      3 wells = well_plate["B2:B7"]
----> 5 await lh.dispense([wells[0]], vols=[15], use_channels=[1])
      7 # for well in well_plate["B2:B7"]:
      8 #   dispenses = await lh.dispense([well], vols=[15], use_channels=[1])

File pylabrobot/pylabrobot/machines/machine.py:25, in need_setup_finished.<locals>.wrapper(self, *args, **kwargs)
     23 if not self.setup_finished:
     24   raise RuntimeError("The setup has not finished. See `setup`.")
---> 25 return await func(self, *args, **kwargs)

File pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:913, in LiquidHandler.dispense(self, resources, vols, use_channels, flow_rates, offsets, liquid_height, blow_out_air_volume, **backend_kwargs)
    911 # liquid(s) for each channel. If volume tracking is disabled, use None as the liquid.
    912 if does_volume_tracking():
--> 913   liquids = [c.get_tip().tracker.get_liquids(top_volume=vol)
    914             for c, vol in zip(self.head.values(), vols)]
    915 else:
    916   liquids = [[(None, vol)] for vol in vols]

File pylabrobot/pylabrobot/resources/tip_tracker.py:62, in TipTracker.get_tip(self)
     55 """ Get the tip. Note that does includes pending operations.
     56 
     57 Raises:
     58   NoTipError: If the tip spot does not have a tip.
     59 """
     61 if self._tip is None:
---> 62   raise NoTipError(f"{self.thing} does not have a tip.")
     63 return self._tip

NoTipError: Channel 0 does not have a tip.
```

</p>
</details> 

I get the same error if I use channel 2 for the dispense (which does not have a tip) but instead get a more reasonable error:

    NoTipError: Channel 2 does not have a tip.

I think that the issue is in how this `zip` is setup, as it does not guarantee that channels are properly matched to volumes:

[image]

rickwierenga, 2024-10-08:
thank you for the detailed report and figuring out what the issue was!

fixed and tested with https://github.com/PyLabRobot/pylabrobot/pull/273

closed 2024-10-08 by rickwierenga

naikymen, 2024-10-08:
Thank you Rick!