PyLabRobot/pylabrobot #274: Cytation 5 Issue Related to libftdi

owen-melville, 2024-10-10:
import matplotlib.pyplot as plt
from pylabrobot.plate_reading import PlateReader, Cytation5Backend
pr = PlateReader(name="PR", size_x=0,size_y=0,size_z=0, backend=Cytation5Backend())

This code has trouble running after installation. The issue has to due with pylibftdi which uses something called libftdi, and the library can't be found. I tried to figure out how to get this library using chatGPT (basically downloading libftdi and adding the folder with .dlls to the path) but it didn't seem to make any difference. 

Has anyone experienced this or have any suggestions for resolving this issue? 

Thanks,
Owen

rickwierenga, 2024-10-10:
libftdi needs to be installed. which os are you using?

owen-melville, 2024-10-10:
windows. I tried to install using chatGPT to help - I downloaded some files and add a folder to the path. I didn't really install anything, is there is a straightforward way to do this?

rickwierenga, 2024-10-10:
it's important to copy both libftdi and libusb dlls. i have used https://www.dependencywalker.com in the past to debug

rickwierenga, 2024-10-10:
if you go into the pylibftdi source code, you can see where it's searching for dlls. Make sure libftdi and libusk live in one of those directories (they should be in the same dir)

owen-melville, 2024-10-10:
oh ok, so transfer the .dlls into the folder where it is looking, I think that makes sense

rickwierenga, 2024-10-22:
how is this going?

owen-melville, 2024-10-24:
Using: " conda install conda-forge::libftdi " worked in my virtual environment
Unfortunately, I'm now getting another error "   pylibftdi._base.FtdiError: b'usb_open() failed' (-4) "
This happened with any program trying to use pylibftdi.
I tried a few things that chatGPT suggested, mostly plugging out/in and trying to install FTDI drivers from their website. Unfortunately I am still getting this error.

rickwierenga, 2024-10-31:
did you find the directory where pylibftdi is searching? did you pust the libftdi and libusb dlls there?

owen-melville, 2024-10-31:
I wasn't sure how to check what directory it was searching. 

WHen i used "conda install conda-forge::libftdi " it did work to stop the error, but I don't know how that command works exactly. maybe I also need to try libusb with " conda install conda-forge::libusb-package " - I will try that

owen-melville, 2024-10-31:
As un update, I tried to install libusb, and it worked successfully but I am still getting the error:
 "   **pylibftdi._base.FtdiError: b'usb_open() failed' (-4)** "

owen-melville, 2024-10-31:
I fixed the error! I tried to replace the drivers in a bunch of ways, but I think it was reverting to the same driver every time. I used a tool called "Zadig" to change the drivers to libusk (based on chatgpt) and that fixed the error.

rickwierenga, 2024-10-31:
Zadig is awesome! It's part of the installation instructions for liquid handling robots (https://docs.pylabrobot.org/user_guide/installation.html#using-the-firmware-interface-with-hamilton-or-tecan-robots), but not for other devices. I shall update those instructions. Thank you for the detailed reporting!

closed 2024-10-31 by rickwierenga

owen-melville, 2024-10-31:
I am very excited to try the tool! I will next try to get the imaging working.

rickwierenga, 2024-10-31:
for imaging there are is an additional program to install (sigh..): these steps worked for me on my macbook and my raspberry pi: https://docs.pylabrobot.org/user_guide/installation.html#cytation5-imager