PyLabRobot/pylabrobot #292: Issue Installation (Spinnaker SDK)

owen-melville, 2024-11-01:
I was trying to install the Spinnaker SDK and I got an error
[image]

This was not resolved by disabling the firewall or running as administrator as suggested by ChatGPT. 

Any suggestions? 

Also - this isn't an issue but I tried out using the plate_read and it was working but I noticed that the function does it for all wells (instead of specific wells) and it does only specified wavelength rather than for a spectra (eg min ---> max). I imagine that I could implement the spectra by just plate reading at different wavelengths, but is there a way to specify the well, rather than doing all of them? I see that this feature was implemented for the camera, and there's a method in the backend that does this. Not sure how simple it would be to do.

rickwierenga, 2024-11-01:
do you have the install log? (I haven't installed it on Windows myself but happy to see if there's anything I may be able to help with.)

---

For reading plates: unfortunately the API is at a whole-plate level. There may be a hack where we say (gaslight) that the plate only has one well in a specific location, but I haven't played with this. I know the machine (plr not yet) supports plates with more or less than 96 wells, so seeing how Gen5 handles that would be my approach.

As for spectra, this is not implemented yet. Do you think we should return a `list of [a list of a list of floats]` (plate data is a `[a list of a list of floats]`)? This could be a nice front-end (PlateReader) feature, which calls the backend (Cytation5, ClarioSTAR, etc.) a number of times at different wavelengths.

rickwierenga, 2024-11-01:
(in an attempt to keep things organized and searchable: please open new issues/[forum](https://discuss.pylabrobot.org) posts for different questions in the future :))

owen-melville, 2024-11-01:
OK for sure - I'm new to using github! 

It's interesting that its plate-level, in the software you can specify what wells you want the plate reading to be done on eg (A1...A3) but also that requires that you have an understanding of what "Plate" is being used. When I used the python wrapper, you have to use a Plate Interface... Unfortunately the wrapper that I worked on only allows you to run experiments that are predefined in the software, which was why I was hoping to move to the API.

rickwierenga, 2024-11-01:
ah, so more may be possible on the firmware level! I shall look into it, hopefully next week.

rickwierenga, 2026-09-02:
we are using #985 now which works much better

closed 2026-09-02 by rickwierenga