PyLabRobot/pylabrobot #305: Issues with pre-commit hooks on Windows

jrast, 2024-11-15:
The pre-commit hooks added in 283eb40 rely on `make` being available, which is mostly not the case on windows.
However, the makefile simply calls ruff which can be done in pre-commit directly: https://github.com/astral-sh/ruff-pre-commit

Also I don't recommed to run the tests on each commit:
- They take far to long for a commit (hang on windows)
- Prevent to work on a PR which still has some issues.
- Maybe there is a pre-commit option to only run the tests on the main branch?

Tests should be run manually and in CI to prevent merges with broken tests.

Typechecking seems to run fast but falls into a simmilar category as tests.

jrast, 2024-11-15:
However, it would make sense to run `ruff check --select I --fix` (sort imports) and `ruff format` to automatically apply a coding guidline and prevent whitespace or reordering changes in commits.

rickwierenga, 2024-11-15:
fixed with https://github.com/PyLabRobot/pylabrobot/commit/26664f90b85cb669029164447d81f6e9cd754aa9

using another external repo just to run ruff in pre-commit seemed excessive, so i just use the cli, hope that works

jrast, 2024-11-18:
When using pre-commit, using the external repos is recommeded as the version of the hook is fixed. The hook environment is automatically shared between different projects (if the same version is used), making the overhead negible. Also the tool is only run on the files which where actually affected by the commit.

jrast, 2024-11-18:
But other than this, the pre-commit hooks now run under windows.  👍

jrast, 2024-11-18:
BTW: I still think that typechecking should not be done during pre-commit. It makes development cumbersome as the typing needs to be correct for every commit. For the moment I removed type checking from the config.

rickwierenga, 2024-11-19:
opened a new branch for you to evaluate changes before merging https://github.com/PyLabRobot/pylabrobot/pull/310

rickwierenga, 2024-11-19:
commits were getting piled up on main so let's just get it as good as we can in 310 and then merge