PyLabRobot/pylabrobot #365: [help] Merge instructions for new git history

naikymen, 2025-01-19:
Hi all!

I've been busty the last few months and just now realized that the commit history has been re-written. Apparently to remove old references to PyHamilton, but I couldn't find an announcement.

The issue is that the commit history is now entirely independent of the one I have :( and I'm not sure how to bring my commits to the new history.

[image]

I imagine that the developers have  already solved this. Would someone be willing to lend me a hand? I've tried some things unsuccessfully.

Thanks in advance!

rickwierenga, 2025-01-19:
yes, i did do this and made a post here: https://discuss.pylabrobot.org/t/fixing-git-lfs-for-plr-required-rewriting-git-history/70/

it should be possible to rebase your changes onto the new main branch.

naikymen, 2025-01-20:
I overestimate my searching skills. Sorry about that, and thanks. :)

naikymen, 2025-01-20:
For the record I ended up making a diff file and applying it to the last common commit. I lost the commit history of my changes but its not really a problem.

closed 2025-01-20 by naikymen

rickwierenga, 2025-01-20:
thank you for helping posterity!