PyLabRobot/pylabrobot #427: liquid handler dispense (lh.dispense) removes liquid from tracker even if action fails

chuandudx, 2025-03-13:
We tried running a script to:
- pick up 200 uL from a well
- dispense 60 uL into another well

We incorrectly set the target well initially, leading to failed dispense actions.

After 3 attempts, PLR's tracker thinks we only have 20 uL left (3x60 uL dispensed) when we haven't dispensed any liquid.

rickwierenga, 2025-03-13:
how did the dispense fail? could you share the specific error? depending on the error, the plunger might have moved down.

chuandudx, 2025-03-13:
The dispense failed with the error: 
First we have a `ChannelizedError(errors={0: PositionNotReachableError...})` when we try to dispense 

Then when we run it again towards the correct well:
- TooLittleLiquidError: Tracker only has 0uL

Our current workaround is set `set_volume_tracking(False)`

Thank you!

rickwierenga, 2025-03-15:
fixed with https://github.com/PyLabRobot/pylabrobot/pull/429. could you please confirm it works for you?

chuandudx, 2025-03-17:
Thank you for the quick fix, Rick! We will verify by the end of this week!