PyLabRobot/pylabrobot #518: OT2 seems to now support manual channel selection

hazlamshamin, 2025-05-23:
Have tested removing all the 'assert use_channels == [0], "manual channel selection not supported on OT for now"' for pick up, aspirate, dispense and drop tip in the file, and the liquid handling works right (PLR installed from source). 

Should this be allowed already? Here is my test code that got no error just by removing those assertions.

```
#!/usr/bin/env python3
import asyncio
import ot_api
import ot_api.requestor as _req
from requests import HTTPError

from pylabrobot.resources.opentrons.deck      import OTDeck
from pylabrobot.resources.opentrons.tip_racks import (
    opentrons_96_tiprack_300ul,
    opentrons_96_filtertiprack_20ul,
)
from pylabrobot.resources.opentrons.plates    import nest_96_wellplate_200ul_flat
from pylabrobot.liquid_handling               import LiquidHandler
from pylabrobot.liquid_handling.backends      import OpentronsBackend

async def main():
    deck   = OTDeck()
    tip300 = opentrons_96_tiprack_300ul(name="tip300")
    tip20  = opentrons_96_filtertiprack_20ul(name="tip20")
    plate  = nest_96_wellplate_200ul_flat(name="plate")
    deck.assign_child_at_slot(tip300, 1)
    deck.assign_child_at_slot(tip20,  2)
    deck.assign_child_at_slot(plate,  3)

    backend = OpentronsBackend(host="192.168.0.188")
    lh      = LiquidHandler(backend=backend, deck=deck)
    await lh.setup()

    # 100 µL with P300 on left
    await lh.pick_up_tips(tip300["A1"], use_channels=[0])
    await lh.aspirate(plate["A1"], vols=[100], use_channels=[0])
    await lh.dispense(plate["A2"], vols=[100], use_channels=[0])
    await lh.return_tips(use_channels=[0])

    # 10 µL with P20 on right
    await lh.pick_up_tips(tip20["A1"], use_channels=[1])
    await lh.aspirate(plate["B1"], vols=[10], use_channels=[1])
    await lh.dispense(plate["B2"], vols=[10], use_channels=[1])
    await lh.return_tips(use_channels=[1])

    await backend.home()
    run_id = ot_api.run_id
    if run_id:
        try:
            _req.post(f"/runs/{run_id}/cancel")
        except HTTPError as err:
            if err.response.status_code == 404:
                _req.post(f"/runs/{run_id}/actions/cancel")
            else:
                raise
        except:
            try:
                _req.delete(f"/runs/{run_id}")
            except:
                pass

    await backend.stop()
    print("✅ Done—homed, cancelled, and disconnected cleanly!")

if __name__ == "__main__":
    asyncio.run(main())

```

rickwierenga, 2025-05-23:
if that works, then yes for sure! do you want to make a PR?

rickwierenga, 2025-05-23:
also should this code you wrote be part of `OpentronsBackend.stop`:

```
run_id = ot_api.run_id
if run_id:
    try:
        _req.post(f"/runs/{run_id}/cancel")
    except HTTPError as err:
        if err.response.status_code == 404:
            _req.post(f"/runs/{run_id}/actions/cancel")
        else:
            raise
    except:
        try:
            _req.delete(f"/runs/{run_id}")
        except:
            pass
```

hazlamshamin, 2025-05-24:
Done created a PR, thanks!

closed 2025-05-24 by hazlamshamin