PyLabRobot/pylabrobot #584: Tip hovers above well when aspirating​dispensing with OpentronsBackend (OT-2)

hazlamshamin, 2025-06-19:
Running a simple “pick up aspirate dispense” script through PyLabRobot moves the pipette to the correct XY but leaves the tip several millimetres above the 96 well plate. The same deck calibration and the equivalent Opentrons Python protocol (run via the official Opentrons app) place the tip inside the well, so the height mismatch seems to come from PyLabRobot. Do I miss something (any argument needed for the function), or is Z-offset always needed for the OpentronsBackend for every labware, despite I have calibrated everything on the official app already?

Attached are the code used for both using the official app vs using pylabrobot, as well as the videos for both execution (https://drive.google.com/drive/folders/1IrGhFlpnYfPBNmx5MvRgfthFThHeONzO?usp=share_link). Disclaimer, not liquid is defined in both approaches.

**Environment**
- Opentrons official app: Version 8.4.1 (8.4.1)
- Robot server version: v8.4.1
- Robot's firmware version: v1.1.0-25e5cea
- Supported Protocol API Versions: v2.0 - v2.23
- Python: 3.10.17
- Pylabrobot: 0.1.6 (installed from source)

**Expected behaviour**
Tip should descend into the liquid, about 1-2 mm just above the bottom of the well, matching the official Opentrons protocol.

**Code used for official Opentrons app**
```
from opentrons import protocol_api

metadata = {
    'protocolName': 'Basic Aliquoting Test',
    'author': 'OpentronsAI',
    'description': 'Basic aliquoting protocol using P300 and P20 single-channel pipettes',
    'source': 'OpentronsAI'
}

requirements = {
    'robotType': 'OT-2',
    'apiLevel': '2.22'
}

def run(protocol: protocol_api.ProtocolContext):
    # Load labware
    tip300 = protocol.load_labware('opentrons_96_tiprack_300ul', 1)
    tip20 = protocol.load_labware('opentrons_96_filtertiprack_20ul', 2)
    plate = protocol.load_labware('nest_96_wellplate_200ul_flat', 3)
    
    # Load pipettes
    p300 = protocol.load_instrument('p300_single_gen2', 'left', tip_racks=[tip300])
    p20 = protocol.load_instrument('p20_single_gen2', 'right', tip_racks=[tip20])
    
    # Protocol steps
    protocol.comment("Starting basic aliquoting protocol")
    
    # Step 1: 100 µL transfer with P300 (left mount)
    protocol.comment("Transferring 100 µL with P300 from A1 to A2")
    p300.pick_up_tip()
    p300.aspirate(100, plate['A1'])
    p300.dispense(100, plate['A2'])
    p300.drop_tip()
    
    # Step 2: 10 µL transfer with P20 (right mount)
    protocol.comment("Transferring 10 µL with P20 from B1 to B2")
    p20.pick_up_tip()
    p20.aspirate(10, plate['B1'])
    p20.dispense(10, plate['B2'])
    p20.drop_tip()
    
    protocol.comment("Protocol completed successfully")
```

**Code used using Pylabrobot**
```
#!/usr/bin/env python3
import asyncio

from pylabrobot.resources.opentrons.deck      import OTDeck
from pylabrobot.resources.opentrons.tip_racks import (
    opentrons_96_tiprack_300ul,
    opentrons_96_filtertiprack_20ul,
)
from pylabrobot.resources.opentrons.plates    import nest_96_wellplate_200ul_flat
from pylabrobot.liquid_handling               import LiquidHandler
from pylabrobot.liquid_handling.backends      import OpentronsBackend

async def main():
    deck   = OTDeck()
    tip300 = opentrons_96_tiprack_300ul(name="tip300")
    tip20  = opentrons_96_filtertiprack_20ul(name="tip20")
    plate  = nest_96_wellplate_200ul_flat(name="plate")
    deck.assign_child_at_slot(tip300, 1)
    deck.assign_child_at_slot(tip20,  2)
    deck.assign_child_at_slot(plate,  3)

    backend = OpentronsBackend(host="192.168.0.188")
    lh      = LiquidHandler(backend=backend, deck=deck)
    await lh.setup()

    # 100 µL with P300 on left
    await lh.pick_up_tips(tip300["A1"], use_channels=[0])
    await lh.aspirate([plate["A1"]], vols=[100], [use_channels=[0])]
    await lh.dispense([plate["A2"]], vols=[100], use_channels=[0])
    await lh.discard_tips(use_channels=[0])

    # 10 µL with P20 on right
    await lh.pick_up_tips(tip20["A1"], use_channels=[1])
    await lh.aspirate([plate["B1"]], vols=[10], use_channels=[1])
    await lh.dispense([plate["B2"]], vols=[10], use_channels=[1])
    await lh.discard_tips(use_channels=[1])

    await backend.home()

    await backend.stop()
    print("✅ Done—homed, cancelled, and disconnected cleanly!")

if __name__ == "__main__":
    asyncio.run(main())
```

rickwierenga, 2025-06-19:
Thank you for the detailed issue with a video and code!

An offset should not be required.

The way the OT backend works is a little complicated compared to other backends. Since the OT server manages its own resource model, we have to mirror the PLR resource model to theirs. Basically, any time a resource is assigned in PLR we send a request to their server. When commands are called (like aspirate), we just refer to the resource identifier (name) and the OT will go there. (This is different from other backends were we can simply say "aspirate at xyz".)

This means that if there is a discrepancy, it could actually be with the way we load resource definitions or upload them to the server, either with the tip rack of plate. I can't immediately see an issue here.

Looking at the [ot_api](https://github.com/rickwierenga/opentrons-python-api/) aspirate command,

```python

@command
def aspirate(
  labware_id: str,
  ...
):
  params = {
    "labwareId": labware_id,
    "wellName": well_name,
    "wellLocation": {
      "origin": "top",
      ...
  }
```

i see the origin is set to "top"

in [the api docs](https://docs.opentrons.com/http/api_reference.html#tag/Maintenance-Run-Management/operation/create_run_command_maintenance_runs__runId__commands_post) you see this refers to the origin of the offset (why would an origin matter for a relative location?). it doesn't make a lot of sense, but it's the only thing i could find

if you changed this in the ot_api (install with -e or just directly in env/lib/...), do you notice a change?

[image]

rickwierenga, 2025-06-20:
oh very nice! that is very confusing in terms of the api documentation, but nice that it works i guess :) i will make a PR for this

rickwierenga, 2025-06-20:
@hazlamshamin please confirm if the new version works expected and if so close the issue :)