PyLabRobot/pylabrobot #588: Confusing line/typo in docs here?

aidan-baydush, 2025-06-27:
https://github.com/PyLabRobot/pylabrobot/blob/7c59632cb1cb3192b904371d40848ee19a8d5d15/docs/user_guide/machine-agnostic-features/writing-robot-agnostic-protocols.md?plain=1#L41

this line did not make sense, looking at the code the strictness level of STRICT (...strictness.Strictness.STRICT) should be the one that raises an error?

Strictness:

https://github.com/PyLabRobot/pylabrobot/blob/7c59632cb1cb3192b904371d40848ee19a8d5d15/pylabrobot/liquid_handling/strictness.py#L5

implementation:
https://github.com/PyLabRobot/pylabrobot/blob/7c59632cb1cb3192b904371d40848ee19a8d5d15/pylabrobot/liquid_handling/liquid_handler.py#L329

aidan-baydush, 2025-06-27:
Is the ```<pylabrobot.liquid_handling.strictness.Strictness.IGNORE>``` correct since that is for the STRICT enumeration? I could just be getting confused.

rickwierenga, 2025-06-28:
i still don't understand what you are saying

rickwierenga, 2025-06-28:
> Is the <pylabrobot.liquid_handling.strictness.Strictness.IGNORE> correct since that is for the STRICT enumeration?

rickwierenga, 2025-06-28:
ohoh the docs are wrong