PyLabRobot/pylabrobot #679: TipTracker Errors on STAR 96-head Using Partially-Filled TipRack

BioCam, 2025-09-24:
This is already working when using STAR_backend directly on the machine but not with Visualizer+tip_tracking:

```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[27], line 4
      1 # Remove supernatant
      2 for idx in range(no_plates):
----> 4     await lh.pick_up_tips96(lh.deck.get_resource("tip_1000ul_00"))

File ~/Desktop/GitHub/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:1396, in LiquidHandler.pick_up_tips96(self, tip_rack, offset, **backend_kwargs)
   1394       tip_spot.tracker.rollback()
   1395     self.head96[i].rollback()
-> 1396   raise error
   1397 else:
   1398   for i, tip_spot in enumerate(tip_rack.get_all_items()):

File ~/Desktop/GitHub/pylabrobot/pylabrobot/liquid_handling/liquid_handler.py:1390, in LiquidHandler.pick_up_tips96(self, tip_rack, offset, **backend_kwargs)
   1388 pickup_operation = PickupTipRack(resource=tip_rack, offset=offset)
   1389 try:
-> 1390   await self.backend.pick_up_tips96(pickup=pickup_operation, **backend_kwargs)
   1391 except Exception as error:
   1392   for i, tip_spot in enumerate(tip_rack.get_all_items()):

File ~/Desktop/GitHub/pylabrobot/pylabrobot/liquid_handling/backends/hamilton/STAR_backend.py:2193, in STARBackend.pick_up_tips96(self, pickup, tip_pickup_method, z_deposit_position, minimum_height_command_end, minimum_traverse_height_at_beginning_of_a_command)
   2191     break
   2192 if prototypical_tip is None:
-> 2193   raise ValueError("No tips found in the tip rack.")
   2194 assert isinstance(prototypical_tip, HamiltonTip), "Tip type must be HamiltonTip."
   2195 ttti = await self.get_or_assign_tip_type_index(prototypical_tip)

ValueError: No tips found in the tip rack.
```

[image]

rickwierenga, 2025-09-24:
is tip tracking enabled when you hit this error case?

BioCam, 2025-09-24:
Yes, and the error disappears when I 
```python
set_tip_tracking(False)
```

BioCam, 2025-09-24:
But when using the STAR directly, I do not use `set_tip_tracking`, the machine doesn't give an error and the Visualizer correctly shows the deck/tip state.

rickwierenga, 2025-09-24:
the problem occurs because somehow all `has_tip` return false

https://github.com/PyLabRobot/pylabrobot/blob/106aef9c8699ceb826d8c9c894eba304a082f24d/pylabrobot/liquid_handling/backends/hamilton/STAR_backend.py#L2187-L2191

BioCam, 2025-09-24:
expected separate cell just above the pick_up_tips96:
[image]

bug hunter printout:

[image]

output during pick_up_tips96:

[image]

😕

BioCam, 2025-09-24:
Where does it remove the tips?

rickwierenga, 2025-09-25:
fixed with https://github.com/PyLabRobot/pylabrobot/pull/680

closed 2025-09-25 by rickwierenga